dmadison/ArduinoXInput #16: incompatible with platformio

Copen-08, 2019-07-15:
Hi, while this library is listed as compatible with platformio it is impossible to actually setup the board profile to allow a board like the Pro Micro to be seen as a XInput controller using platformio (along with their only officially supported ide -  Visual Studio Code). It seems like having a platform.json file in the https://github.com/dmadison/ArduinoXInput_SparkFun would fix this incompatibility.

dmadison, 2019-07-16:
Hey there! Thanks for making the issue.

Where did you read that this library was compatible with PlatformIO? I don't use PlatformIO myself and I'm sure I didn't mark it as compatible anywhere. That may be an automated thing, as the library is separate from the various board backend packages.

Could you clarify why the boards files are incompatible? Is there some reason the installation instructions for the boards files don't work with PlatformIO?

dmadison, 2019-07-30:
It's been two weeks with no response. Closing due to not enough information.

closed 2019-07-30 by dmadison