dmadison/ArduinoXInput #17: ATmega16u2

jtodaaaa, 2019-07-17:
Hi,
I'm thinking of using ATmega16u2 for my small controller which I hope would works like an xbox360 controller with only one switch and tow led's.
Could you give me suggestions about which files I would need to modify in your XInput library ?
Thank you.

dmadison, 2019-07-17:
You shouldn't need to modify any files in the library. You will however need to modify [the AVR core files](https://github.com/dmadison/ArduinoXInput_AVR) to provide a 'board' definition for the 16U2.

jtodaaaa, 2019-07-17:
Thank you.
I will try.

jtodaaaa, 2019-07-20:
Hi,
ATmega16u2 has only two timers, 8bit and 16bit,  and four USB endpoints.
Is this OK to use with XInput library ?
Thank you.

dmadison, 2019-07-20:
There library doesn't use any timers (it uses the USB SOF interrupt for timing events), but the 4 endpoints are a problem. Out of the box I rewrote the AVR core to use configurable endpoints 1, 2, and 5 - purposefully not allocating the other endpoints provided in the config descriptor. The number 5 endpoint is necessary otherwise the lack of ACK/NACK will cause the PC driver to drop the controller.

It might be possible to rewrite the config descriptor to assign the endpoints 'out of order' so that you can allocate the memory for the three endpoints in use, but I don't know if the PC driver would play nice (I've never tried it myself). You could also try rewriting the code to share In/Out endpoint memory by toggling the endpoint type to 'in' and back to 'out' every time you need to send a packet, but that's more involved and risks missing packets.

---

For what it's worth, the 32U4 works out of the box with no modifications required anywhere. If you can stomach the extra cost it would save you some time.

jtodaaaa, 2019-07-20:
Thank you for the explanation.
I already tried 32U4, and it works fine.
I think I should follow your advice.
Modification for 16u2 seems to be beyond my ability.
Thanks again.

dmadison, 2019-07-20:
Good luck! Closing this issue for now.

closed 2019-07-20 by dmadison