dmadison/ArduinoXInput #19: Window 10

nyd5178, 2019-07-25:
My Teensy is not recognized on Window 10

In Xbox 360 Controller for Windows property,

"This device can not be started. (Code 10)

There is not enough system resources to complete the API."

Board : Teensy LC
USB Type : Xinput
CPU Speed : 48 MHz
Optimize : Smallest Code
Programmer : AVRISP mkll

Do not speak English well.. Thank you for your help

dmadison, 2019-07-25:
Hello there!

Please provide some more information:

* What version of Arduino are you running?
* What version of Teensyduino are you running?
* What version of ArduinoXInput_Teensy (the Teensy board files) did you use?
* How long is your USB cable? Have you tried a shorter one?
* Is your USB cable connected directly to a USB port on the computer's motherboard, or through a hub?
* What are the specifications for your computer? Processor name, amount of memory (RAM), etc.
* Are you able to reprogram the Teensy by using the 'reset' button on top?

nyd5178, 2019-07-25:
Previously we used Arduino1.8.8 and Teensyduino1.45

But the latest file you just uploaded did not have any errors!

Thankyou!

dmadison, 2019-07-25:
Good, I'm glad you were able to get it working!

nyd5178, 2019-07-26:
Sorry, last but one more question. How can my PC do Serial communication with Teensy? The language is cpp.

dmadison, 2019-07-26:
Unfortunately you cannot use the USB serial communication simultaneously as XInput. The project emulates an Xbox 360 wired controller, which does not have serial communication.

nyd5178, 2019-07-29:
It's a sad news for my project ...
If so, do you think there's a way to use the Xbox 360 Wireless Controller?

dmadison, 2019-07-29:
I'm not sure what you mean.

This project emulates a wired controller. The wireless controller functions differently and would be an entirely different project to try and emulate. Neither controller uses serial communication.

If you need USB serial communication, you would need to use a second microcontroller and send commands to and from the XInput microcontroller.

nyd5178, 2019-07-29:
I'm sorry. I did not understand how it works.....
I just wanted to use it on an Android OS.

dmadison, 2019-07-29:
I've never tried it, but this project should work on Android. Although, again, without USB serial.

If you need to emulate a gamepad with USB serial you can try DirectInput (e.g. [the Arduino Joystick library](https://github.com/MHeironimus/ArduinoJoystickLibrary)). It's an open standard but it's much older, and many modern games don't support it.

closed 2019-08-11 by dmadison