dmadison/ArduinoXInput #20: Does this work with actual Xbox 360 consoles?

Jdbye, 2019-07-27:
The reason I want to know is because I want to stream games for a friend so he can try out a 360 game I've been showing him without needing to buy a 360.

dmadison, 2019-07-27:
No it does not. Genuine controllers have a hardware security chip on the PCB whose security method has not been (openly) broken. The PC driver does not check for this chip but the console does, which means unauthorized clones like this library won't work with it.

The lack of console compatibility is stated in the `README` files for the various boards packages, but I do realize that it wasn't stated anywhere in this repository so I've added a disclaimer to the `README`.

Jdbye, 2019-07-28:
I see, thanks. You just saved me from wasting money buying a Teensy. That leaves me only one option (that I know of) that's supposed to work, and that's a ConsoleTuner Titan One, which enables authentication by piggybacking on a real 360 controller and can be controlled from a PC. It's twice as expensive as a Teensy, and needs a wired 360 controller (which I don't have), but at least it should work.
I wonder how 3rd party controllers do it? Do they buy the special sauce from Microsoft to get it to work? What about cheapo chinese controllers, surely they didn't buy security chips off of M$ to make it work. I feel like they've figured out a way around it, and the open source community hasn't caught on yet. But then again, they have the money to pay a team to reverse engineer the controller and figure out what makes it tick -- not so for the open source community, so I can't expect that kind of thing.

dmadison, 2019-07-29:
3rd party controllers buy licensing rights from Microsoft for permission to make controllers for the console. I don't know how cheapo Chinese controllers do it, but I'd assume there's some gray market sales of the security chips going on.

Eventually I'd like to enable the library to catch and pass along the authentication requests so you can use a genuine controller to authenticate with the console, but I have no plans for that at the moment.

closed 2019-08-11 by dmadison

gdsports, 2020-09-02:
There is an easy way to use an Arduino XInput controller with a console. Use a controller converter. Brook and MayFlash make converters in the $20 to $50 range. For example, a Teensy 3.6 running Arduino XInput connected to a MayFlash Magic S Pro connected to a PS4 works for me. There are more expensive converters that can handle all of the major consoles.

Build one Arduino XInput custom controller and use it on all consoles supported by converters.