dmadison/ArduinoXInput #23: Adafruit ItsyBitsy/other boards?

Stetoskop, 2019-08-05:
Hey, I took a look at the API and really couldn't wrap my head around working it out for Adafruit boards (ItsyBitsy in my case). Is this board planned to be added anytime soon?

dmadison, 2019-08-05:
At the moment I don't have any plans to add support for additional boards, no.

Which ItsyBitsy board? The 32U4 one is Adafruit's version of the 5V SparkFun "Pro Micro", and can probably be programmed using that board profile with no issues. You could also try using the default "Leonardo" board type. If you want to add support for the board yourself, I recommend looking at the SparkFun XInput board repository and trying to copy those changes over to the Adafruit AVR repository - it *should* be fairly straight-forward. You need to redirect each board to use the XInput AVR core and then change the VID / PID to match the Xbox controller.

The M0 and M4 ItsyBitsy boards are more complicated. They use a variant of the SAMD Arduino core, which I have never modified to work with XInput. You would need to make similar changes to the ones in the XInput AVR core (removing pluggable USB and CDC serial, setting a receive interrupt, allocating endpoint memory, etc.), but tailored to SAMD.

Stetoskop, 2019-08-08:
Yeah my bad, I somehow just messed up my board option multiple times when uploading. The "Arduino/Genuino Micro w/XInput" to be exact, worked flawlessly for the Adafruit ItsyBitsy 32u4 5V.

Will keep in mind what you wrote when I get around to other *actually* unsupported boards. Thanks!

closed 2019-08-08 by Stetoskop