dmadison/ArduinoXInput #25: Emulate multiple XInput controllers

sarcus, 2019-10-09:
Hey,

As far as I could see, the library only allows to emulate one controller at a time.
Is it techically possibile and how hard would it be to emulate more? Do you have any hints how to achieve it?

Thanks

dmadison, 2019-10-10:
It's technically possible but you'd essentially need to recreate much of the project from scratch. The current library is built around emulating the wired controller. To emulate multiple controllers on one microcontroller you'd need to emulate the Xbox 360 wireless receiver, which is a different sort of beast.

To start, you'd need to reverse engineer how the wireless receiver works. Figure out which USB endpoints correspond to which controllers, how to enable/disable different controllers, how data packets are formatted, and determine what (if any) communication is going on in the background.

You'd then need to build the customized USB stack for the microcontroller of your choice. I'd imagine that the 32U4 (Leonardo, Pro Micro, etc.) is out because it only has the hardware for a handful of USB endpoints, but I haven't looked at the wireless receiver's descriptors myself so that's just a hunch.

Lastly, you'd need to modify the library backend API to support multiple controllers. At that point the library would only need a constructor modification (e.g. "XInput(uint8_t controller_number)") and then you could create different objects for each controller. In theory, at least.

---

In short, it would be a lot of work. It's not impossible, mind you, but I have no interest in doing it. Especially since you can get the same result by adding another $2 microcontroller.

sarcus, 2019-10-10:
Ok, thanks. I thought it could work by emulating USB Hub, but now I see on the web that it is not so easy.

$2 microcontroller? How would you do that? I thought second Arduino board is required, which is quite more expensive.

dmadison, 2019-10-10:
You could either use a second Arduino board or just the bare 32U4 microcontroller with the supporting electronics. 'Knockoff' Arduino Pro Micros are around $2 from Chinese wholesalers (AliExpress and the like).

sarcus, 2019-10-19:
Ok, I see. Thank you.

closed 2019-10-19 by sarcus