dmadison/ArduinoXInput #28: Using 'get rumble' to send a high level signal to a relay

g-coded, 2019-12-24:
Is this possible?  I managed to set a pin to low, but can't get it to go high when testing the rumble.  I'm not too familiar with Arduino coding so I may just be setting this up wrong.

dmadison, 2019-12-24:
Take a look at the tutorial here: https://www.partsnotincluded.com/tutorials/how-to-emulate-an-xbox-controller-with-arduino-xinput/

This should be all that you need:

```cpp
	// Get rumble value
	uint16_t rumble = XInput.getRumble();
 
	// If controller is rumbling, turn on LED
	if (rumble > 0) {
		digitalWrite(Pin_LED, HIGH);
	}
	else {
		digitalWrite(Pin_LED, LOW);
	}
```

g-coded, 2019-12-24:
I used that as a template, and modified the:
 
// Output Pins
const uint8_t Pin_LED = LED_BUILTIN;
 to 
// Output Pins
const uint8 Pin_Rumble = 12; 
and then changed the "If controller is rumbling" to:
digitalWrite(Pin_Rumble, HIGH);
	}
	else {
		digitalWrite(Pin_Rumble, LOW);
	}

I'm not sure if I can just name the pin, and define the "if rumble >0 digital write to a digital pin?

dmadison, 2019-12-24:
Have you read through that tutorial, or the source for the rumble functionality? I'm happy to answer simple questions, but most of this should be explained elsewhere in greater detail.

There are three rumble functions: 
```cpp
XInput.getRumbleLeft();  // 8 bit, big motor
XInput.getRumbleRight();  // 8 bit, small motor
XInput.getRumble();  // 16 bit, left motor as 0xFF00, right motor as 0x00FF
```

There are two motors, both of them with an 8-bit analog range. You can use either or both. The library provides the data, how you set up the output (digital or analog) is up to you.

g-coded, 2019-12-25:
Thanks @dmadison , I should have seen the 0-255 range, since this was just oulined to control an led I thought it may be limited to on/off.

I looked through the other examples and found an 'analog write' in ReceiveCallback.ino:

// If we have a rumble packet (0x00), see our rumble data on the LED
	else if (packetType == (uint8_t) XInputReceiveType::Rumble) {
		uint8_t rumbleValue = XInput.getRumbleLeft() | XInput.getRumbleRight();
		analogWrite(LED_Pin, rumbleValue);

My plan is to use the analog signal to control a potentiometer on a motor controller, I'm not sure how to define the output range/value(is analogWrite 0-5v?).  I'm not seeing 'analogWrite/rumbleValue' in the Xinput.h file.

dmadison, 2019-12-25:
It's an example; it shows just one way you could possibly use the data. The left/right rumble functions return an 8-bit unsigned integer (0-255). How you use that information is entirely up to you.

The Arduino `analogWrite()` function takes an 8-bit signed integer as well. The voltage range it maps to depends on the microcontroller and the voltage input. It's also a bit of a misnomer since it doesn't actually output a steady voltage, but pulse-width modulation of the high and low voltages. Whether that works for your specific hardware setup depends very much on how your hardware is designed to work.

---
That being said, this issue tracker is for reporting problems with the library or creating feature requests. It's not meant for asking for my personal help with other's people's projects. As I said I'm happy to answer simple questions about how the library works, but unless you have a problem with the library itself I'll ask you to close this issue.

closed 2019-12-26 by g-coded