dmadison/ArduinoXInput #30: Reprogramming

Synless, 2020-04-05:
Hi,
Is there a way to reprogram the board via USB a second time in /w Xinput config or do I have to use an external programmer ?

dmadison, 2020-04-05:
For AVR boards with the Caterina bootloader, reset the board twice quickly to re-enable serial programming over USB. You can reset the board by grounding the `RESET` pin.

For the Teensy 3 boards you can reprogram the board normally, i.e. you only need to press the `RESET` button once.

[Please read the board-specific documentation](https://github.com/dmadison/ArduinoXInput_AVR#how-to-upload).

Synless, 2020-04-06:
Double reset doesn't work on a 32u4. Device manager detect the serial port for a couple of second after reset, then it disconnect and Xinput shows up. During the brief moment of serial port, I cannot manage to program it.

dmadison, 2020-04-06:
At the risk of repeating myself, have you read the upload instructions in the ArduinoXInput_AVR repository?

Synless, 2020-04-07:
My bad, i haven't noticed there was two different repositories.
It is working fine now. Thanks.