dmadison/ArduinoXInput #32: Bluetooth support?

wolfshow, 2020-04-13:
Is it possbile to support bluetooth board such as hc-05 or esp32?

dmadison, 2020-04-13:
Theoretically, yes. Someone would need to reverse engineer the Xbox One controller and figure out how to get the ESP to emulate it, then build an API wrapper to integrate with the library.

I am not going to add wireless support myself. I have no experience developing with Bluetooth and I don't have to hardware to experiment with it.

closed 2020-04-26 by dmadison

Rmdhn1, 2022-01-10:
Theoretically it is, but needs a usb dongle, use a compatible board as a usb dongle and another sending the values

Rmdhn1, 2022-01-10:
If you wanna be cheeky look into controller emulation softwares and send HID data, as in keypresses and mouse movements, some even allow them to differentiate between "keyboards" and "mouse" so you can still use your main keyboard while using the emulated Arduino one