dmadison/ArduinoXInput #34: Working on some android phones, but not for all

wolfshow, 2020-06-13:
I have successfully build a gamepad for my android phones, I tested xiaomi, LG, razer phones. It works on LG and xiaomi 5, but not xiaomi 6 and razer phone 2. Is that because the android version?

dmadison, 2020-06-13:
It's impossible to say without seeing logs of the USB data.

dmadison, 2020-06-14:
Sorry, I should clarify: the library emulates the Xbox controller but does so imperfectly. This is partially due to the closed source nature of the specification and partially due to the hardware limitations of the Arduino boards. But because it's mimicking other hardware that's expected to behave a certain way there's little to do in the way of debugging - if the driver accepts the board everything works, if it doesn't everything breaks. The only way I'm aware of to find out what's causing the issue is to inspect the data on the USB lines directly with a logic analyzer or USB analyzer.

I'll also add that if this issue likely lies with the specific board files you're using and not with the library itself. Please duplicate this issue on the respective boards repository you've tested with.

closed 2020-06-27 by dmadison