dmadison/ArduinoXInput #39: Joystick and XInput in same board, but not at the same time.

Dionysusnu, 2020-08-18:
I've got a project in which I would like my Teensy 3.6 to switch between acting as an XInput device and a regular Teensy HID Joystick. Unfortunately, this is not possible with the board provided by the library. The difference with [this issue](https://github.com/dmadison/ArduinoXInput/issues/22) is that only one of the two options will be active at once. Is that possible with some changes to the library, or is it a USB limitation?

dmadison, 2020-08-18:
Unfortunately the Teensy core was never designed to be modular like that. The 'standard' answer is that if you need multiple USB modes you would compile with one of the composite USB device modes. XInput throws a wrench in that solution because it doesn't work as a composite device.

You would have to rewrite the Teensy core USB implementation to allow runtime switching between USB modes (descriptors, endpoints, and interrupts). It's possible but it wouldn't be an easy task.

Just to be clear: no USB implementation is done in this library, period. All USB issues and USB-related problems are the scope of the modified 'boards' files. The library is just a wrapper for the pre-existing data packets. If you want to implement this, those are the files you will need to modify.

Dionysusnu, 2020-08-18:
Ok, that's a shame.

closed 2020-08-18 by Dionysusnu