dmadison/ArduinoXInput #40: Joystick drift

drcrazymonkey7, 2020-09-22:
I have a full set of buttons and 2 joysticks to emulate a full xbox controller for use with a raspberry pi. To test the buttons, I'm plugging it into my windows computer. Every button works but the right joystick does not. The right joystick has a constant drift, while the left works fine. Even with the right joystick unplugged from the Teensy LC board, the drift continues. Any ideas on what might be causing this drift?

dmadison, 2020-09-22:
How are you observing / measuring the drift? If you use [gamepad-tester.com](https://gamepad-tester.com/) what do the joysticks read? Have you tried swapping the left and right joysticks in the code to see if the problem is hardware related?

It should drift while it's unplugged, as that leaves the analog input floating.

drcrazymonkey7, 2020-09-22:
Thank you for replying. I have attached a quick video of the gampad-tester.com readings, it seems that both joysticks are reading wrong. 
These are the joysticks I am using: https://www.amazon.com/gp/product/B07SJYWW8J/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1
Each joystick has a 5v and GND wire connected to the Teensy LC boards GND and 5v pins. 
This is the readings from the gampad-tester: https://drive.google.com/file/d/1Cf25tOTNEpSvkRyM3WSaVMbVFqycTidJ/view?usp=sharing

Thank you again for taking your time to help.

dmadison, 2020-09-22:
>Each joystick has a 5v and GND wire connected to the Teensy LC boards GND and 5v pins.

This is your problem. The Teensy is a 3.3V device and is not tolerant of 5V on any of its inputs - you need to connect to 3.3V not 5V.

Joysticks are potentiometers are heart, and when you connect all three pins (high voltage, wiper, and ground) you create a simple voltage divider. Centered, they produce ~2.5V (5V/2) which is about 75% of a 3.3V max or about 50% past center (1.65V), which is what you see on gamepad-tester.com.

Note that because the Teensy LC is not 5V tolerant it is possible that you have permanently damaged the Teensy's inputs by connecting to the 5V pin, in which case you will need to replace the microcontroller.

drcrazymonkey7, 2020-09-22:
Thank you so much, I rewired it to the 3.3v pins, and everything is working fine now. Thank you for taking the time to help me!

closed 2020-09-22 by dmadison