dmadison/ArduinoXInput #41: psv analog stick support

wolfshow, 2020-09-28:
with this xinput driver, the X-axis and Y-axis of psv analog stick are always between 0 to 60, not 0-100. Is there anything I can do to make the stick working from 0 to 100?

dmadison, 2020-09-28:
The library is hardware agnostic, meaning that it doesn't care what you're connecting to the outputs so long as the data is valid. If the analog outputs do not go to their max you're not feeding them data that hits the max of the range.

By default the range is `-32768` to `32767` (int16_t) but the library also supports automatic rescaling with the `setJoystickRange` function (min/max). See the [GamepadPins](../blob/master/examples/GamepadPins/GamepadPins.ino) example for a demonstration using a 10-bit ADC.

Check your hardware, check your wiring, or change your code to work with a different range of inputs.

wolfshow, 2020-09-28:
Thanks for the quick reply!

closed 2020-09-28 by wolfshow