dmadison/ArduinoXInput #48: Arduino Leonardo port not found under port tab, but seen as a controller on device page.

3L3CTR0STAT1C, 2021-02-07:
I wanted to test out the code, so I uploaded it to the device, and when I tried to open the serial monitor, it did not open. then I selected the port and it was not there. So I proceeded to reset the bard and unplug it, but it still did not work. I even tried burning the bootloader with a UNO as ISP. but no luck.

and as I said, I cant see it in ports, but I see it on the device manager page in settings as a controller.

I believe it is because of the library, because, there was another issue on the Arduino forums about the problem, with the Xinput. So please help.

dmadison, 2021-02-07:
Please, *please* read the documentation :worried:. It's there for a reason.

[Here is the README for the AVR boards files](https://github.com/dmadison/ArduinoXInput_AVR/blob/master/README.md). Please read the ["Upload Warning and Instructions" section](https://github.com/dmadison/ArduinoXInput_AVR/blob/master/README.md#upload-warning-and-instructions), which starts off with "**!!!!!!! IMPORTANT !!!!!!!**" and ends with "**Do *not* upload XInput sketches to your Arduino unless you know how to reset it**".

>I even tried burning the bootloader with a UNO as ISP. but no luck.

Burning the bootloader does not erase the XInput sketch. You have to change the program memory, i.e. use the "Upload Using Programmer" option in the IDE or use `avrdude` with a compiled hex file.

3L3CTR0STAT1C, 2021-02-08:
Thank you for your information, I was in a hurry and did not read it. Now you wrote, "Do not upload XInput sketches to your Arduino unless you know how to reset it! Otherwise, you will not be able to program it anymore and you'll have to reflash the bootloader with an external programmer." so will I have to do the quick reflash, or can I use the method you mentioned about the rest button. 

Also, if the computer can't see it as a port, how can I access the serial port? it is not mandatory in my project, but I need to look at the values before processing.

3L3CTR0STAT1C, 2021-02-08:
I just attempted and it told me done uploading, thank you for your help, but how can open the serial monitor?

dmadison, 2021-02-08:
You can't. The project emulates an Xbox controller. The Xbox controller has no serial port, so there is no serial port on the Arduino when it's pretending to be an Xbox controller.

If you want to look at the values on the controller you will either need to use the library without the XInput board (i.e. upload w/ the regular Leonardo selected) or connect a second board and send those values with the hardware UART / I²C / SPI bus.

3L3CTR0STAT1C, 2021-02-08:
Ok thank  you ill just use the adafruit libraty with a small display. thank you for your help!

closed 2021-02-08 by 3L3CTR0STAT1C