dmadison/ArduinoXInput #52: Change recommended test app

offalynne, 2021-05-20:
html5gamepad.com has been loaded with sketchy adware and autoplay ads and is apparently advertising sale of scraped user data

suggest pointing users to https://greggman.github.io/html5-gamepad-test/ or https://apps.nektro.net/gamepad

dmadison, 2021-05-27:
Thanks for the heads-up, that's unfortunate to hear.

Do you know of an alternatives that share the same user-friendly interface? Both of your suggested alternatives work but they lack the graphic that shows which inputs correspond to which axis/button on the controller. I think that's important since users are not working with physical controllers and it can help identify swapped or mis-assigned inputs.

Previously the README directed users towards [gamepadviewer.com](https://gamepadviewer.com/), which has a nice display of the controller but lacks the debug information from html5gamepad.

closed 2022-12-21 by dmadison

offalynne, 2022-12-21:
In my opinion pointing to either of the above (along with gamepadviewer.com for visualizer if that actually is necessary) is much better than pointing traffic toward a site that is [a transparent data-gathering scheme](https://gamepad-tester.com/privacy):
> _For a better experience while using our Service, we may require you to provide us with certain personally identifiable information, including but not limited to your name, phone number, and postal address._

dmadison, 2022-12-21:
As I said, I am more than happy to change the recommended testing link if you know of an alternative with the same feature set.