dmadison/ArduinoXInput #53: Stop emulation

dorgis16, 2021-05-27:
Hello I’m a bigin

dmadison, 2021-05-27:
Unplugging the microcontroller tends to work pretty well :stuck_out_tongue:

What are you trying to accomplish?

dorgis16, 2021-05-27:
In trying to run multiple emulator like xinput and keyboard but I want to be able to switch between them in the keyboard emulation you can start it with keyboard.begin() and stopped it with keyboard.end() but in xinput I only se a way to start

dmadison, 2021-05-27:
Unfortunately you can't do that with this library. The USB descriptors are handled by the relevant 'boards' files and cannot be changed or overwritten on the fly.

For that matter you can't do that with the official Keyboard library either. [The `begin` and `end` functions are empty](https://github.com/arduino-libraries/Keyboard/blob/baf975294b489a5b6a11f60c2a22df68a92312d8/src/Keyboard.cpp#L69). To truly 'stop' emulation you would have to disconnect from USB (physically or in software) and then re-enumerate without the keyboard descriptors present.

If you want keyboard and joystick behavior to switch off your best bet is to use [the Joystick library](https://github.com/MHeironimus/ArduinoJoystickLibrary) and some sort of XInput emulation on the PC side (e.g. x360ce) if it's needed. Although the Arduino will always show up as both a keyboard and joystick simultaneously.

If you need the device to actually switch from one to the other you would have to leave the Arduino environment. You could use [LUFA](http://www.fourwalledcubicle.com/LUFA.php) with an AVR Arduino (e.g. the Leonardo or Pro Micro) but you would be on your own with coding the USB descriptors and behavior. And if you're a beginner programmer that's going to be a tall task.

dorgis16, 2021-05-28:
Thank for all your help I really appreciate

closed 2021-05-28 by dmadison