dmadison/ArduinoXInput #54: 2 key 1 function

themarine1147, 2021-06-16:
I am making a arcade stick for all games(not only fighting games)
I am trying to put one function(like L3) to two different keys
but It does not work, If I write code like that, one key became disable...
Is there anyway to do this?

dmadison, 2021-06-16:
You have to 'or' the logic so that either key can enable it. It's the coding equivalent of putting buttons in parallel.

E.g.:

```cpp
bool button1 = !digitalRead(x);  // '!' assuming pull-ups
bool button2 = !digitalRead(y);

bool buttonPressed = button1 || button2;  // '||' = 'or'
```

themarine1147, 2021-06-17:
I have coded a script below for 22 keys xinput joystick. because of lack of pins, I used matrix pin 8,9,10(input), pin 11,12,13(input), 3x3 matrix.  I was successful to make working joystick, but I tried to use matrix[8][13] and matrix[10][13] to function  as BUTTON_L3

Could you do some advice to how to use "or" fucntion?

#include <XInput.h>

// Setup
const boolean UseLeftJoystick   = true;  // set to true to enable left joystick
const boolean InvertLeftYAxis   = false;  // set to true to use inverted left joy Y

const boolean UseRightJoystick  = true;  // set to true to enable right joystick
const boolean InvertRightYAxis  = false;  // set to true to use inverted right joy Y

const boolean UseTriggerButtons = true;   // set to false if using analog triggers

  
const int ADC_Max = 1023;  // 10 bit

//rumble

// Joystick Pin
const int Pin_stickUp = 2;
const int Pin_stickRight = 3;
const int Pin_stickDown = 4;
const int Pin_stickLeft = 5;

const int Pin_RightJoyX = A0;
const int Pin_RightJoyY = A1;

// Trigger Pins
const int Pin_TriggerL = 6;
const int Pin_TriggerR = 7;

// Button Pins
const int Pin_ButtonA = 0;
const int Pin_ButtonB = 1;
const int Pin_ButtonX = A2;
const int Pin_ButtonY = A3;

const int Pin_ButtonLB = A4;
const int Pin_ButtonRB = A5;

// Directional Pad Pins

void setup() {
	// If using buttons for the triggers, use internal pull-up resistors
	if (UseTriggerButtons == true) {
		pinMode(Pin_TriggerL, INPUT_PULLUP);
		pinMode(Pin_TriggerR, INPUT_PULLUP);
	}

	// Set buttons as inputs, using internal pull-up resistors
  pinMode(Pin_stickUp, INPUT_PULLUP);
  pinMode(Pin_stickRight, INPUT_PULLUP);
  pinMode(Pin_stickDown, INPUT_PULLUP);
  pinMode(Pin_stickLeft, INPUT_PULLUP);

	pinMode(Pin_ButtonA, INPUT_PULLUP);
	pinMode(Pin_ButtonB, INPUT_PULLUP);
	pinMode(Pin_ButtonX, INPUT_PULLUP);
	pinMode(Pin_ButtonY, INPUT_PULLUP);

	pinMode(Pin_ButtonLB, INPUT_PULLUP);
	pinMode(Pin_ButtonRB, INPUT_PULLUP);

  pinMode(8, INPUT_PULLUP);
  pinMode(9, INPUT_PULLUP);
	pinMode(10, INPUT_PULLUP);
	pinMode(11, OUTPUT);
  pinMode(12, OUTPUT);
  pinMode(13, OUTPUT);

keyPressed()

	XInput.setJoystickRange(0, ADC_Max);  // Set joystick range to the ADC
	XInput.setAutoSend(false);  // Wait for all controls before sending

	XInput.begin();
}

void loop() {

	// Read pin values and store in variables
	// (Note the "!" to invert the state, because LOW = pressed)

  
  boolean stickUp = !digitalRead(Pin_stickUp);
  boolean stickDown = !digitalRead(Pin_stickDown);
  boolean stickLeft = !digitalRead(Pin_stickLeft);
  boolean stickRight = !digitalRead(Pin_stickRight);

	boolean buttonA = !digitalRead(Pin_ButtonA);
	boolean buttonB = !digitalRead(Pin_ButtonB);
	boolean buttonX = !digitalRead(Pin_ButtonX);
	boolean buttonY = !digitalRead(Pin_ButtonY);

	boolean buttonLB = !digitalRead(Pin_ButtonLB);
	boolean buttonRB = !digitalRead(Pin_ButtonRB);

	// Set XInput buttons
	XInput.setButton(BUTTON_A, buttonA);
	XInput.setButton(BUTTON_B, buttonB);
	XInput.setButton(BUTTON_X, buttonX);
	XInput.setButton(BUTTON_Y, buttonY);

	XInput.setButton(BUTTON_LB, buttonLB);
	XInput.setButton(BUTTON_RB, buttonRB);

//matrix

digitalWrite(11, LOW);
if (digitalRead(8)) XInput.release(DPAD_LEFT);
else XInput.press(DPAD_LEFT);
if (digitalRead(9)) XInput.release(DPAD_UP);
else XInput.press(DPAD_UP);
if (digitalRead(10)) XInput.release(BUTTON_START);
else XInput.press(BUTTON_START);
digitalWrite(11, HIGH);

digitalWrite(12, LOW);
if (digitalRead(8)) XInput.release(DPAD_RIGHT);
else XInput.press(DPAD_RIGHT);
if (digitalRead(9)) XInput.release(DPAD_DOWN);
else XInput.press(DPAD_DOWN);
if (digitalRead(10)) XInput.release(BUTTON_BACK);
else XInput.press(BUTTON_BACK) ;
digitalWrite(12, HIGH);

digitalWrite(13, LOW);
if (digitalRead(8)) XInput.release(BUTTON_L3);
else XInput.press(BUTTON_L3); 
if(digitalRead(9)) XInput.release(BUTTON_R3);
else XInput.press(BUTTON_R3); 
if (digitalRead(10)) XInput.release(BUTTON_L3);
else XInput.press(BUTTON_L3);
digitalWrite(13, HIGH);

	// Set XInput trigger values
	if (UseTriggerButtons == true) {
		// Read trigger buttons
		boolean triggerLeft  = !digitalRead(Pin_TriggerL);
		boolean triggerRight = !digitalRead(Pin_TriggerR);

		// Set the triggers as if they were buttons
		XInput.setButton(TRIGGER_LEFT, triggerLeft);
		XInput.setButton(TRIGGER_RIGHT, triggerRight);
	}
	// Set left joystick
  

	if (UseLeftJoystick == true) {
	int stickUp = !digitalRead(Pin_stickUp);
	int stickDown = !digitalRead(Pin_stickDown);
  int stickLeft = !digitalRead(Pin_stickLeft);
  int stickRight = !digitalRead(Pin_stickRight);

 XInput.setJoystick(JOY_LEFT, stickUp, stickDown, stickLeft, stickRight);
	}

	// Set right joystick
	if (UseRightJoystick == true) {
		int rightJoyX = analogRead(Pin_RightJoyX);
		int rightJoyY = analogRead(Pin_RightJoyY);

		boolean invert = !InvertRightYAxis;

		XInput.setJoystickX(JOY_RIGHT, rightJoyX, invert);
		XInput.setJoystickY(JOY_RIGHT, rightJoyY, invert);
	}

  // Get rumble value

	// Send control data to the computer
	XInput.send();
}

dmadison, 2021-06-17:
Try using the snippet I posted above.

themarine1147, 2021-06-17:
Thank you I got it!

closed 2021-06-17 by themarine1147