dmadison/ArduinoXInput #55: Please move the USB_XINPUT checks to the header file

Rayzeq, 2021-09-01:
I suggest you move the USB_XINPUT checks from XInput.cpp (line 29 to 55) to XInput.h, because I want to use it on an Arduino Uno R3, so I created a firmware for the atmega16u2 and a .h file that implements the XInputUSB api. I would like to use it like this:
```c++
#include "MyHeader.h" // define USB_XINPUT
#include <XInput.h>
```
But right now it doesn't work because XInput.cpp doesn't take the MyHeader.h file into account.

Rayzeq, 2021-09-01:
After doing it on my local files, I just realized that it doesn't really work, but I think there should be a way to create an XInputUSB api without modifying the Arduino installation files.

dmadison, 2021-09-02:
As you discovered, the preprocessor checks can't be moved to the header because the `.ino` and the library `.cpp` are in two separate compilation units. The preprocessor checks don't carry over.

I agree with you that it would be nice not to have to modify the Arduino installation files but I can't think of a good way to do it. This is mostly a quirk with how the Arduino build system is so rigid. If you want to modify the "makefile" you need to modify the Arduino core. The library and the API are designed to function independent of one another, so the API must be defined elsewhere in the core in order for the library to see it.

If you can think of a clean solution I'd be happy to implement it.

Rayzeq, 2021-09-02:
I couldn't find a solution either, so I added `#include "MyHeader.h"` in XInput.h, it's not a good solution, but it works.

closed 2021-09-02 by Rayzeq