dmadison/ArduinoXInput #58: PlatformIO build_flags

rootexpression, 2021-11-18:
It is in PlatformIO's list of libraries but the build flags seem a mystery.  I've tried
build_flags = -D X_INPUT
USB_XINPUT
XINPUT
USB_APIXINPUT
XInputUSB_API
etc etc (the last 2 are what it offered for autocomplete so I rolled with it)

Either refuses to compile (USB_XINPUT, which seems to be the sensible option), or throws no errors but lsusb shows the teensy as a serial device.

Is this just a no-go or is there something I'm missing?

`
  return XInputUSB::connected();
         ^
.pio/libdeps/teensy31/XInput/src/XInput.cpp: In member function 'int XInputController::send()':
.pio/libdeps/teensy31/XInput/src/XInput.cpp:444:9: error: 'XInputUSB' has not been declared
  return XInputUSB::send(tx, sizeof(tx));
         ^
.pio/libdeps/teensy31/XInput/src/XInput.cpp: In member function 'int XInputController::receive()':
.pio/libdeps/teensy31/XInput/src/XInput.cpp:453:6: error: 'XInputUSB' has not been declared
  if (XInputUSB::available() == 0) {
      ^
.pio/libdeps/teensy31/XInput/src/XInput.cpp:459:24: error: 'XInputUSB' has not been declared
  const int bytesRecv = XInputUSB::recv(rx, sizeof(rx));
                        ^
Compiling .pio/build/teensy31/FrameworkArduino/HardwareSerial1.cpp.o
*** [.pio/build/teensy31/libe15/XInput/XInput.cpp.o] Error 1
========================== [FAILED] Took 1.78 seconds ==========================
 `

dmadison, 2021-11-19:
Have you followed the installation instructions outlined in the README?

rootexpression, 2021-11-20:
I did, and I installed the `Teensy XInput USB Mode` package per its instructions.

dmadison, 2021-11-20:
The build flag to activate the XInput descriptors on Teensy is `USB_XINPUT`. Like the other USB build flags it's mutually exclusive. I do not use PlatformIO myself, but [the documentation says that](https://docs.platformio.org/en/latest/platforms/teensy.html):

>A default macro is set to -D USB_SERIAL if no one is specified.

It looks like this is handled in a [preprocessor script](https://github.com/platformio/platform-teensy/blob/6319aab241ad9a526de55c013934595aa89299dd/builder/frameworks/arduino.py#L40) that enumerates all of the USB types. My guess is that you would need to modify the script to add the missing flag so that PlatformIO doesn't create the serial define (which takes precedence in the preprocessor statement order).

closed 2021-12-30 by dmadison