dmadison/ArduinoXInput #59: Can XINPUT Receive Serial1 Command?

Kuantronic, 2021-12-26:
I'm using an Arduino Leonardo (ATMega32U4). My test code is designed to receive serial command from the HW Serial1 port and translate it to Xbox controller inputs. However when Serial1 commands are sent, no controller inputs is detected by the [Gamepad Tester](https://gamepad-tester.com/). With my troubleshooting right now, I'm looking at the possibility that the XINPUT library and board manager prevents Serial1 from working. Thus, it's not properly receiving and interpreting the Serial1 commands.

Would anyone be able confirm my theory or know more about it?

dmadison, 2021-12-26:
The XInput modifications shouldn't affect the hardware UART, and I'm afraid I can't replicate your issue. Try this simple demo program, running a jumper between pins 0 (RX) and 1 (TX):

```cpp
auto & iface = Serial1;

bool state = 0;
unsigned long last = 0;

void setup() {
	iface.begin(115200);
}

void loop() {
	unsigned long now = millis();
	if(now - last >= 1000) {
		last = now;
		state = !state;
		iface.write(state);
	}

	int c = iface.read();
	if(c != -1) {
		digitalWrite(13, c);
	}
}
```

Using both the standard ("Arduino Leonardo") and XInput ("Arduino Leonardo w/ XInput") cores, the onboard LED should blink slowly.

Kuantronic, 2021-12-27:
Hi Dave,

I got it to work after modifying the code to repeat back what it received in the hardware UART. Not sure why it worked but it worked. Thanks for your clarification that it shouldn't affect HW UART though!

closed 2021-12-27 by Kuantronic