dmadison/ArduinoXInput #61: I'm getting double (ghost) presses sometimes.

liam-maps, 2022-05-19:
Sometimes when using my arcade sticks have double presses on directions, which is kinda inconvenient.
I can notice it specially browsing through EmulationStation menus.
I'm using Arduino Pro Micro.

I'm aware of other libraries based on DInput that allow for a debounce, but this library doesn't seems to have that option.
Could it be implemented?
Thank you for this amazing library.

dmadison, 2022-05-19:
Hi there. I purposefully did not add a debouncing implementation because it adds significant bloat and gets in the way of directly controlling the output data. If your inputs need to be debounced there are a number of existing libraries that do an excellent job and can be easily integrated with ArduinoXInput. I've used [Bounce2](https://github.com/thomasfredericks/Bounce2) in the past with good results.

Are you using the `GamepadPins` example by any chance?

liam-maps, 2022-05-22:
Hi!
Thanks for your reply.
Yes, I'm using the GamepadPins example with minor changes (pin reassignment to my cabling layout, basically)
Is it wrong that I use that sketch? Is there a better way?
I'll look onto Bounce2 to see if I can figure it out.

Thank you again.

dmadison, 2022-05-22:
It's not "wrong" per-se, but it was never intended to be a complete solution. It was originally supposed to be a straightforward example of how to use the library with minimal parts on hand and minimal code (hence the lack of debouncing, input structs, loops, and other quality of life perks). Unfortunately I think I went too far by including all of the pins, which has led to many people using it in their projects unmodified rather than as a reference to write their own code.

I'm hoping to get around to reworking the examples with the next library update, which will hopefully avoid these sorts of problems going forwards.

Re: Bounce 2, it's reasonably straight-forward to implement:

```cpp
#include <Bounce2.h>
#include <XInput.h>

#define PIN 2
Bounce2::Button button;

void setup() {
    button.attach(PIN, INPUT_PULLUP);
    button.interval(30);  // debounce time, ms
    button.setPressedState(LOW);  // 'LOW' is pressed, 'HIGH' is released

    XInput.begin();
}

void loop() {
    button.update();

    if(button.changed()) {
        XInput.setButton(BUTTON_A, button.isPressed());
    }
}
```

(Untested, but that's the gist).

I'm not sure of your skill level, but for multiple buttons I would probably create a struct containing the button ID, pin number, and Bounce2 object so that you can iterate through them.

closed 2022-12-21 by dmadison