dmadison/ArduinoXInput #62: Toggle left/right analog stick

crowcey, 2022-06-06:
New to all this, greatly appreciate your code and all your hard work. 

Is there a way to use a switch to toggle a single joystick between left or right stick.

dmadison, 2022-06-06:
Hi there. You can store the joystick ID in a variable, then change it as needed. Something like:

```cpp
XInputControl joystick;

if(digitalRead(SwitchPin) == LOW) {
    joystick = JOY_LEFT;
}
else {
    joystick = JOY_RIGHT;
}
```

Then use the standard joystick 'set' function and pass the `joystick` variable as the ID.

crowcey, 2022-06-07:
Thank you very much. Think that makes sense.

crowcey, 2023-03-07:
Can you do the same with the push buttons. I've tried every combination but can't wrap my head round it.

dmadison, 2023-03-07:
Yes, in exactly the same way:

```cpp
XInputControl button;

if(digitalRead(SwitchPin) == LOW) {
    button = BUTTON_A;
}
else {
    button = BUTTON_B;
}

XInput.setButton(button, state);
```

crowcey, 2023-03-07:
BINGO!!! Thank you very much, I was close but just making a few blunders. Thanks for your help