dmadison/ArduinoXInput #63: iOS compatibility

Rennerdo30, 2022-06-11:
I would like to build a gamepad for my iphone, but sadly it does not detect the arduino when set to xinput.
Could there be a fix for this, or is it something like #57? I'm not sure how apple determines if it is a genuine xbox gamepad...
Maybe with some tweaks it could fall into the MFi (Made For iPhone) category.

https://support.apple.com/en-us/HT210414
https://support.apple.com/en-us/HT211232

dmadison, 2022-06-12:
Unfortunately I do not have a modern iOS device, so I have no way of knowing what the issue is.

My guess is that the iOS driver does not properly enumerate the endpoints from the descriptor, and instead assumes the endpoints based off of the VID/PID alone. Due to how the AVR architecture is set up, the Arduino implementation tweaks the endpoint numbers so that it doesn't need to share hardware registers between input and output. It's theoretically possible to fix that but it requires some juggling of the endpoint data that didn't seem worthwhile at the time.

It's also possible that iOS is trying to use some features of the controller that are not supported by the Arduino implementation, such as audio I/O. If that's true the Arduino won't even NACK the packets correctly and the USB host will shut down the connection.

Of course, that's all theoretical. You would have to attach a logic analyzer to the USB data lines to see exactly what's going wrong. If you can figure out what's going on and engineer a fix I'm more than happy to merge it, but I'm afraid I won't have the time to work on this anytime soon.

Rennerdo30, 2022-06-12:
Thanks for your fast response.
I'm sadly not very familiar with this stuff. 
But since I'm thinking about reverse engineering the Nintendo Joy Con Protocol, buying  logic analyzer seems not too far off.
I'll give it a try and update this issue if get some results.

dmadison, 2022-12-21:
Closing as stale.

closed 2022-12-21 by dmadison