dmadison/ArduinoXInput #65: Question : Use this package without access to Arduino Reset Button?

joshcalebrobinson, 2022-08-06:
Hello,

I'm using an [LBX](https://junkfoodarcades.com/products/lbx), a Leonardo Arduino based controller, I bought online. I was trying to use this package to get the controller to pretend it's an Xbox Controller; but I don't have access to the Arduino reset button.

I read in the [ArduinoXInput_AVR repo](https://github.com/dmadison/ArduinoXInput_AVR#how-to-upload) that I can't upload INO files unless I have access to the reset button. But in this repo, the [ArduinoXInput repo](https://github.com/dmadison/ArduinoXInput#getting-started) nothing implies that I do need to reset the Arduino during the upload process. Are there wrappers in this package that means I no longer have to reset the Arduino during the upload?

dmadison, 2022-08-06:
Hello!

There is no mention of the upload process in this repo because there is nothing in this library that affects the upload process. The files in the AVR core modify how uploading works for the AVR boards in particular, and that is why the directions are there.

The reset button is required because of how the uploading process works for AVR. The XInput core removes the serial interface that is used to normally send the reset signal for upload, so the user needs to do it manually. You can also use an external programmer to upload via the ICSP pins if they are exposed.

I looked briefly and although the product page says it's open source I could not find any gerbers or board files, so unfortunately I can't help there. You may need to contact the manufacturer to find out what pins are exposed on the PCB (or open it up and poke around).

joshcalebrobinson, 2022-08-06:
I found out I can actually open it up and reset it by bridging the ground and reset pins. 

[image]

I got the upload to work once (and PC recognized controller as Xbox 360), but after that couldn't get flash to work. 

```
Uploading using selected port: COM4
C:\Users\joshu\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\joshu\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf -v -patmega32u4 -cavr109 -PCOM4 -b57600 -D -Uflash:w:C:\Users\joshu\AppData\Local\Temp\arduino_build_460990/LBX-XInput.ino.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\joshu\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"

         Using Port                    : COM4
         Using Programmer              : avr109
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega32U4
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : butterfly
         Description     : Atmel AppNote AVR109 Boot Loader

Connecting to programmer: .An error occurred while uploading the sketch
```

dmadison, 2022-08-06:
> I got the upload to work once

The first time you upload there is no XInput code on the microcontroller and the upload proceeds normally. Subsequent uploads have the XInput code on the board and require doing a timed reset.

I'm afraid there's not much I can do to help. You will have to follow the upload instructions in the AVR core README. If you're having trouble, you can always solder a header on and flash using an external programmer.

joshcalebrobinson, 2022-08-06:
Ah ok, that makes sense. 

I'll probably figure out how to whip up an external reset button and continue messing around with this XInput repo for a bit.

To be clear, if I can hook up an external reset button I could just re-upload my original non-XInput INO file using a timed reset to restore my board to its original functionality? I'm pretty new to this Arduino stuff.

dmadison, 2022-08-06:
Yep, exactly. The XInput code makes no permanent changes to the microcontroller itself, it's just another firmware (albeit with a few quirks).

joshcalebrobinson, 2022-08-06:
Gotcha, thanks!

Tl;dr for anyone who views this after the fact:
**You need to be able to reset the Arduino to use this package.**

joshcalebrobinson, 2022-08-06:
Follow up, I was able to reupload INO files on the [LBX](https://junkfoodarcades.com/products/lbx) by hot wiring the ground and reset pins from above.

closed 2022-08-06 by joshcalebrobinson