dmadison/ArduinoXInput #68: not detected + reset not working?

Rmdhn1, 2022-09-10:
so right now, my micro "successfully" flashes without pressing the reset twice 
"avrdude done.  Thank you."

but it isn't detected by windows, and when it's uploading it doesn't stop, i'd used to flash another board and on the uploading it stops and that's where you double click reset, now this one, it doesn't. if i do double click either it has an error or does nothing.

Rmdhn1, 2022-09-10:
device manager still detects it as micro also says
"Device USB\VID_2341&PID_8037&MI_00\7&2c5b71e9&0&0000 requires further installation."
dk what to do

Rmdhn1, 2022-09-10:
another attempt leads to this :
[image]
sadly still did nothing as on device manager still detected as arduino micro with the vid pid

Rmdhn1, 2022-09-10:
should i try to change the vid pid manually?

dmadison, 2022-09-10:
Hello. Are you using the XInput AVR core? Make sure you're selecting the "with XInput" version of the board in the IDE.

If so, please enable verbose uploading in the Arduino IDE options window and post the full message when attempting to upload. The "avrdude done" message will display even when nothing has been uploaded to the board.

Regardless of the XInput setup, if your board is showing a flag in the device manager you will need to [install or update the drivers](https://docs.arduino.cc/tutorials/generic/DriverInstallation) for best compatibility with the IDE.

Rmdhn1, 2022-09-10:
oh yea forgot to use the xinput board, thanks,

closed 2022-09-22 by Rmdhn1