dmadison/ArduinoXInput #69: Will there be support for ESP-32 ?

Ugobyte, 2022-09-23:
I have an Espressif ESP-32 that basically runs Arduino library and code. 
Is there a way to make this library work for an ESP-32? I tried adding your library to my ESP-32 project but there was #preprocessor #error because my ESP-32 board isn't one of the boards you support.

dmadison, 2022-09-23:
As far as I know the ESP32 does not have a USB interface. Excluding hacks like V-USB, it would be impossible to add support for the chipset.

closed 2022-09-23 by dmadison

Ugobyte, 2022-09-23:
I'm using the ESP-32 Devkit with the USB connector if that makes any difference?

dmadison, 2022-09-23:
Unfortunately it doesn't. USB is complicated, and the ESP32 does not have the hardware in its silicon to "speak" USB. Development boards with a USB port have a secondary IC which converts USB to UART (serial) for communicating with the microcontroller. ([See the diagram here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-devkitm-1-v1.html)).

dmadison, 2022-09-23:
No, as far as I'm aware no custom USB protocols of any sort are possible on the ESP32. The ESP32 (the chip, not the board) does not have any USB capabilities.

ShikyC, 2024-01-04:
> No, as far as I'm aware no custom USB protocols of any sort are possible on the ESP32. The ESP32 (the chip, not the board) does not have any USB capabilities.

To clarify this, ESP32-S2 and ESP32-S3 chips have built in USB (and HID) libraries so in theory porting this repository to ESP32-S2/S3 is possible (but of course, takes effort).