dmadison/ArduinoXInput #77: XInput.h not found

henryfoster, 2022-11-28:
Error message: 

```
C:\Users\andre\Blink\Blink.ino:31:10: fatal error: XInput.h: No such file or directory
 #include <XInput.h>
          ^~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: XInput.h: No such file or directory
```

Code (selected from examples):
```
#include <XInput.h>

void setup() {
	XInput.begin();
}

void loop() {
	XInput.press(BUTTON_A);
	delay(1000);

	XInput.release(BUTTON_A);
	delay(1000);
}

```

Installed https://github.com/dmadison/ArduinoXInput_AVR
installed 1.2.6
selected as board Arduino Micro w/ XInput

dmadison, 2022-11-28:
Hello. That error indicates that the library is not installed correctly. Do you see the library listed in the "examples" dropdown in the IDE?

dmadison, 2022-12-21:
More information needed, closing as stale.

closed 2022-12-21 by dmadison

henryfoster, 2023-01-10:
Hi, sorry for the silence. 

> Do you see the library listed in the "examples" dropdown in the IDE?

Yes I did. 

The problem seems to be with Windows11. I switched to a second PC with Win10 and it worked flawless.