dmadison/ArduinoXInput #80: BUTTON_LOGO Does Not Result in Any Button Press

tcaivano, 2023-01-20:
Using XInput.setButton() with BUTTON_LOGO as the first argument does not do anything.
To test, I first used known working code to confirm my hardware configuration was correct: 
` XInput.setButton(BUTTON_X, setX);`
Then, without changing any other configuration, I replaced this call with the following:
` XInput.setButton(BUTTON_LOGO, setX);`
I would expect this to result in the same behavior as the Xbox button or home button being pressed on a standard controller, but I am unable to observe any button presses using online testers on PC or on Nintendo Switch via an XInput adapter.

dmadison, 2023-01-20:
Hello. Sorry to hear that you're having issues. I've just tested with the latest version of the library and cannot replicate your problem.

You mentioned changing existing code with a hardware configuration. Try this minimal automated example and see if you get the same results:

```cpp
#include <XInput.h>

void setup() {
	pinMode(0, INPUT_PULLUP);
	while(digitalRead(0) == LOW);
}

void loop() {
	XInput.setButton(BUTTON_LOGO, 1);
	delay(1000);
	XInput.setButton(BUTTON_LOGO, 0);
	delay(1000);
}
```

Please test with both the standard XInput setup, and with the board in "debug" mode using the non-XInput board type. Does the button work? Is there a discrepancy between the two modes?

tcaivano, 2023-01-20:
Can confirm that the provided test code works in both modes. Thanks for the confirmation and the quick feedback!

closed 2023-01-20 by tcaivano