dmadison/ArduinoXInput #82: Non-debug mode in PlatformIO IDE

CyberDNIWE, 2023-03-21:
I was able to install and run both debug (printout) and non-debug versions of this lib using the instructions for Arduino IDE (1.x.x).
However, the lib didnt seem to work in non-debug mode when installed (and used) with PlatformIO IDE and its respective library manager. 

Upon flashing xinput's blink example, the board remains visible as COM-port and outputs debug info into serial.
I suspect that the cause is the abcense of appropriate platform installaition, as it was a mandatory step for Arduino IDE, but not completely sure.

Do you have any insights on this issue or maybe I'm missing something crucial for using this lib with PIO?

My `platformio.ini` is as follows:
```ini
[env:sparkfun_promicro16]
platform = atmelavr
board = sparkfun_promicro16
framework = arduino
build_type = release
lib_deps = dmadison/XInput@^1.2.6
```
and my board is `Arduino Pro Micro (clone) 3.3V`, that seemed to be running this library fine when programmed in Arduino IDE with Board selection of: `XInput AVR Boards -> Arduino Micro w/ XInput` that was previously installed from the board manager with added custom json provided in readme.

_I'm also not sure how PIO library management is set up, but the link lead to this repo, so this place is my best bet._

dmadison, 2023-03-21:
Hello, sorry to hear you're having issues.

Officially there is no PlatformIO support. PlatformIO pulls from the Arduino library list which is why this library is listed as compatible, although it probably shouldn't be. Unfortunately I don't use PlatformIO myself so I'm not 100% sure how it likes to be set up.

I believe you may need to fork their AVR platform specification and then change the core target from within the Python build script, although that's just a hunch. The board-specific definitions should stay the same.

CyberDNIWE, 2023-03-21:
Thank you for speedy reply, 
I suspected PIO does the scraping automatically.

closed 2023-03-21 by CyberDNIWE