dmadison/ArduinoXInput #83: A question about Xinput device identification

1781741527, 2023-04-04:
Hello, I like your project very much, it has brought me great convenience in my small project, but now I have encountered a problem, I want to make a steering wheel simulator, but after I imported your project, I can't recognize in the game that I am currently connected to the steering wheel simulator, just recognized as an ordinary controller, I checked some relevant information, I think it should be related to HID protocol or USB identifier, can you give me some suggestions or help, thank you very much!

dmadison, 2023-04-04:
Hello. I'm not quite sure I understand your issue. Are you trying to build a steering wheel device?

This library emulates an Xbox 360 gamepad. It does not support any other device types. Most steering wheels use generic HID descriptors. You may have better luck using a generic HID library such as [the Arduino Joystick library](https://github.com/MHeironimus/ArduinoJoystickLibrary). If you need finer control over the outputs you may need to read the USB HID specification and write your own descriptor.

1781741527, 2023-04-04:
Thank you very much for your reply, this helps me a lot, I really want to make a steering wheel device, since the game I play is Horizon 5, I have an XBox controller and its protocol is XInput, so I am worried about whether this game recognizes the normal control device protocol. Based on your suggestion, I think I would choose another way to implement it. Thanks again for your advice.

closed 2023-04-04 by dmadison