dmadison/ArduinoXInput #85: error: recipe.preproc.macros pattern is missing

strunx, 2023-04-18:
hi,

to get xinput working on TeensyLC i have done following:

1. clean install Arduino IDE 1.8.19
2. install Teensyduino 1.57
3. start Arduino IDE 1.8.19 and install ArduinoXInput library (currently compatible with Teensyduino Version 1.57)
4. get the special ArduinoXInput_Teensy files (Version 1.1.1 to make it compatible with Tennsyduino Version 1.57)
5. delete the vanilla "teensy"-folder in C:\Program Files (x86)\Arduino\hardware and 
6. replace it with the "teensy"-folder from special ArduinoXInput_Teensy files
7. start arduino IDE 1.8.19 and set Board "Teensy LC" with USB-type "xinput" and when compiling
8. this error-massage is displayed:  "recipe.preproc.macros pattern is missing" error compiling

What i am doing wrong?

Thx for help.

dmadison, 2023-04-18:
>delete the vanilla "teensy"-folder in C:\Program Files (x86)\Arduino\hardware

This is wrong. Do not delete the folder, you need to copy the XInput modifications over the existing files.

strunx, 2023-04-18:
hi Dave,

 

thank you for your quick reply.
so i have to put the individual files into the correspondent
folder, one file by one .... and not the whole folder...

 

best regards

 

strunx

 

 

 

Gesendet: Dienstag, 18. April 2023 um 21:25 Uhr

dmadison, 2023-04-18:
You can copy the entire folder, you just need to replace the existing files.

The ArduinoXInput_Teensy repository only contains the handful of files in the core that need to be replaced. From your description it sounds like you are deleting the rest of the core files preemptively.

strunx, 2023-04-19:
all works now. Thank you!

closed 2023-04-19 by strunx