dmadison/ArduinoXInput #87: Is it possible to send additional 'keypresses' or buttons outside of what's provided?

Nixon-Joseph, 2023-06-18:
Scenario: My brother and I are working on a custom controller project - mostly for funzies - and I found your library. So far it's working great, I've generated an input matrix and it's working like a charm.

The problem I'm running into - is I need extra buttons. Like 8 extra buttons. I totally get if they can't be sent as XBox button events or whatever, so I'm cool if it's just an additional keyboard key-press event. I've looked into the Keyboard library, and it seems like what I want, but it doesn't seem compatible when in 'xinput' mode or whatever.

Please let me know if you have any suggestions, or if I'm just totally wrong.

I'm using a Sparkfun Pro Micro.

Here's my current programming:

```c++
#include <XInput.h>

byte rows[] = {4,5,6,7,8};
const int rowCount = sizeof(rows)/sizeof(rows[0]);

byte cols[] = {0,1,2,3};
const int colCount = sizeof(cols)/sizeof(cols[0]);

byte keys[colCount][rowCount];

int l2Pin = 9;
int r2Pin = 10;
int r2PinValue = 0;
int lStickXPin = 20;
int lStickYPin = 21;
int rStickXPin = 18;
int rStickYPin = 19;
int l3Pin = 16;
int r3Pin = 14;

void setup() {
	XInput.begin();
  Keyboard.begin();
  // Serial.begin(9600);
  XInput.setTriggerRange(0, 1024);
  XInput.setJoystickRange(0, 1024);

	for(int x=0; x<rowCount; x++) {
		// Serial.print(rows[x]); Serial.println(" as input");
		pinMode(rows[x], INPUT);
	}

	for (int x=0; x<colCount; x++) {
		// Serial.print(cols[x]); Serial.println(" as input-pullup");
		pinMode(cols[x], INPUT_PULLUP);
	}
}

void readMatrix() {
	// iterate the columns
	for (int colIndex=0; colIndex < colCount; colIndex++) {
		// col: set to output to low
		byte curCol = cols[colIndex];
		pinMode(curCol, OUTPUT);
		digitalWrite(curCol, LOW);

		// row: interate through the rows
		for (int rowIndex=0; rowIndex < rowCount; rowIndex++) {
			byte rowCol = rows[rowIndex];
			pinMode(rowCol, INPUT_PULLUP);
			keys[colIndex][rowIndex] = digitalRead(rowCol);
			pinMode(rowCol, INPUT);
		}
		// disable the column
		pinMode(curCol, INPUT);
	}
}

// /*
// For some reason col 1 and col 2 seem to be swapped.
// Can't find any reason for this ATM.

// ------| col 1 | col 2 | col 3 | col 4 |
// ---------------------------------------
// Row 1 |   Y   |   X   |   B   |   A   |
// ---------------------------------------
// Row 2 |   L1  |   R1  | Start | Select|
// ---------------------------------------
// Row 3 | Misc4 | Misc3 | Misc2 | Misc1 |
// ---------------------------------------
// Row 4 | Right |  Down |   Up  |  Left |
// ---------------------------------------
// Row 5 |   C1  |   C2  |   L4  |   R4  |
// */

void broadcastMatrix() {
  bool isPressed = false;
  int button = -1;
	for (int rowIndex=0; rowIndex < rowCount; rowIndex++) {
		for (int colIndex=0; colIndex < colCount; colIndex++) {
      isPressed = keys[colIndex][rowIndex] == LOW;
      switch (rowIndex) {
        case 0: // Row 1
          switch (colIndex) {
            case 0: // Y
              button = BUTTON_Y; break;
            case 1: // X
              button = BUTTON_X; break;
            case 2: // B
              button = BUTTON_B; break;
            case 3: // A
              button = BUTTON_A; break;
          } break;
        case 1: // Row 2
          switch (colIndex) {
            case 0: // L1
              button = BUTTON_LB; break;
            case 1: // R1
              button = BUTTON_RB; break;
            case 2: // Start
              button = BUTTON_START; break;
            case 3: // Select
              button = BUTTON_BACK; break;
          } break;
        // case 2: // Row 3
          // No mapping for XBox controller
          // switch (colIndex) {
          //   case 0: // Misc 4
          //     button = BUTTON_L1; break;
          //   case 1: // Misc 3
          //     button = BUTTON_R1; break;
          //   case 2: // Misc 2
          //     button = BUTTON_START; break;
          //   case 3: // Misc 1
          //     button = BUTTON_BACK; break;
          // } break;
        case 3: // Row 4
          switch (colIndex) {
            case 0: // Right
              button = DPAD_RIGHT; break;
            case 1: // Down
              button = DPAD_DOWN; break;
            case 2: // Up
              button = DPAD_UP; break;
            case 3: // Left
              button = DPAD_LEFT; break;
          } break;
        // case 4: // Row 4
          // switch (colIndex) {
            // case 0: // C1
            //   button = 0; break;
            // case 1: // C2
            //   button = 0; break;  
            // case 2: // L4 button
              // button = ...; break;
            // case 3: // R4 button
              // button = ...; break;
          // } break;
      }
      if (isPressed) {
        XInput.press(button);
      } else {
        XInput.release(button);
      }
		}	
	}
}

void readAndBroadcastStick(int btnPin, int xPin, int yPin, int joyId) {
  int xBtn = joyId == JOY_LEFT ? BUTTON_L3 : BUTTON_R3;
  pinMode(btnPin, INPUT_PULLUP);
  int btnValue = digitalRead(btnPin);
  pinMode(btnPin, INPUT);
  if (btnValue == LOW) {
    XInput.press(xBtn);
  } else {
    XInput.release(xBtn);
  }
  int stickXValue = analogRead(xPin);
  XInput.setJoystickX(joyId, stickXValue);
  int stickYValue = analogRead(yPin);
  XInput.setJoystickY(joyId, 1024 - stickYValue);
}

void readAndBroadcastTriggers() {
  int l2PinValue = analogRead(l2Pin);
  XInput.setTrigger(TRIGGER_LEFT, l2PinValue);
  int r2PinValue = analogRead(r2Pin);
  XInput.setTrigger(TRIGGER_RIGHT, r2PinValue);
}

void loop() {
	readMatrix();
  broadcastMatrix();
  // Left stick
  readAndBroadcastStick(l3Pin, lStickXPin, lStickYPin, JOY_LEFT);
  // Right stick
  readAndBroadcastStick(r3Pin, rStickXPin, rStickYPin, JOY_RIGHT);
  readAndBroadcastTriggers();
}
```

dmadison, 2023-06-18:
Unfortunately not. The library emulates an existing Xbox gamepad and the driver won't detect it if you add additional interfaces to the USB descriptor.

closed 2023-06-18 by dmadison

Nixon-Joseph, 2023-06-18:
Ahh - thanks for the quick response - Given my scenario, what would you do to accommodate the extra buttons?

I have an 8BitDo pro controller, and it has several modes depending on button held at startup, assuming that's a custom bootloader. One of the options is 'xbox' mode - I don't believe it has any extra buttons - so I'm not trying to say that it does.

Would you perhaps recommend a similar path, where it boots different programs? Obviously that would mean my extra buttons would be useless in 'xbox' mode, but would still be available in other cases.

I'd be happy to receive any additional advice you have - if any!

Thanks!

dmadison, 2023-06-18:
The simplest thing to do would be to use [the Arduino Joystick Library](https://github.com/MHeironimus/ArduinoJoystickLibrary), which uses class compliant HID descriptors. You could set it up to use more buttons than are on the Xbox controller, or add the Keyboard library alongside it without issue.

Although I understand that isn't compatible with many modern games, which is why this library exists in the first place 😄 Though you could always pair the device with x360ce to get around that.

>Would you perhaps recommend a similar path, where it boots different programs?

You could do that, though you would have to leave the Arduino environment to accomplish it. The Arduino USB implementation isn't set up to swap USB modes on the fly.

Nixon-Joseph, 2023-06-18:
Ahh noted -that makes sense - thanks a bunch!

dmadison, 2023-06-23:
As far as I know it's not possible to do from the USB side. "Rumble" isn't a standardized feature of the HID class, and is implemented in the device-specific driver.  It isn't possible for the USB device to say "I have rumble motors for X and Y" in a way that the host understands unless it already knows what the device is.

Nixon-Joseph, 2023-06-23:
Ahh I see. That's unfortunate. Thank you!

I was hoping I could set up some sort of "listener" (serial.read seemed promising) so that I could map the signal from whatever game software (emulators mostly). But you're right, it would have to be specifically tuned to the controller, and I can't imagine that's an option to configure.

My brother will be bummed, but he either loses a bunch of buttons, or rumble. I bet he'll pick to drop rumble 😆.

Thanks again, and nice job on this library. Just wish there were some way to register as xinput device, and also send other keys hahaha. I know that restriction isn't from what you put together.