dmadison/ArduinoXInput #89: Is there a way to have two ArduinoXinput devices and have them show up as two Joysticks?

amaraklov, 2023-08-26:
I was trying this, and it's possible one of the cables I used doesn't provide data, I have to check, but I didn't see anything on hooking up two of these and having windows recognize them as player 1 and player 2 and keep them like that once the hardware id has been registered in the registry.

dmadison, 2023-08-26:
There should be no issues with using multiple ArduinoXInput devices on the same host.

>having windows recognize them as player 1 and player 2 and keep them like that

That isn't up to the controller, the host driver is in charge of assigning player numbers. If you need a microcontroller to be a specific "player", you need to make sure that it enumerates with the host in the proper order.

amaraklov, 2023-08-26:
Ok Thanks for getting back to me

closed 2023-08-26 by amaraklov