dmadison/ArduinoXInput #97: multi-controller support

Berghopper, 2024-08-29:
Related to #91 

It's possible to emulate multiple xbox controllers over 1 usb device using the wireless receiver packets.
I tried to do this myself, and got some basic poc working [here.](https://github.com/Berghopper/360-w-raw-gadget)
However, I did it with a rpi zero, which is easily able to support all the endpoints.

The wireless receiver supports 4 devices, each with 2 interfaces (one control, one for audio) which each have 2 endpoints (in/out).
With my poc I'm not fully able to emulate the receiver, instead I changed the interface protocol for controller input to that of the [regular controller.](https://github.com/Berghopper/360-w-raw-gadget/commit/95e413c387c381ecec6881a737038fc78b02d775)

Maybe it'd also be possible to edit the config descriptor and then e.g., drop the audio interfaces as well. This way the `ATmega32U4` might possibly be able to support 3 controllers. Seeing that this lib aims for that hardware.

I think it'd be awesome if this gets added, seeing that it could make some projects more compact using only 1 device.

Thoughts?

Berghopper, 2024-08-30:
update: indeed, it seems that if you just leave out the audio interface/endpoints, it still works!

dmadison, 2024-11-04:
Sorry for the late reply. That's interesting! Did you have any luck testing with only 6 endpoints?

Berghopper, 2024-11-07:
I did, anywhere between 1-4 controllers seem to work just fine. Main issue is going beyond that, and it's heavily depend on the OS's drivers. e.g. MacOS can't seem to pick up more than 1 controller