dmadison/ArduinoXInput #101: Cannot find the XInput version of Teensy boards

mrebersv, 2025-06-07:
I'm trying to get this going on a Teensy 4.0 using the GamepadPins example on a clean Arduino 2.3.6 install (only version of Arduino IDE installed).  This is Windows 11.  Teensyduino (1.59) works as expected as well.  

I believe I installed ArduinoXInput correctly (just added the library as a zip).  I've since followed the additional instructions for ArduinoXInput_Teensy, I believe.  My `%AppData%\..\Local\Arduino15\packages\teensy\hardware\avr\1.59\` is definitely populated (screenshot1.png).

[image]

So I hop into the cores folder from there, and copy the `teensy3` and `teensy4` folders into `%AppData%\..\Local\Arduino15\packages\teensy\hardware\avr\1.59\cores`.  Some files are overwritten, as expected.  I start up the Arduino IDE, but here's where I seem to falter.  

When I open up the Arduino IDE, and use the example GamepadPins or Blink sketch, I can compile it and send it to my Teensy 4.0, but I don't seem to have an option to send it to `Teensy 4.0 w/XInput` or something similar.  Thus, I can open the serial monitor and monitor button presses, but obviously that means it isn't acting as an XInput device.  

I'm guessing I'm missing something absolutely obvious here, but I can't figure it out for the life of me.  Does anything stand out as wrong in the above?

mrebersv, 2025-06-07:
I meant to add that when compiling, I'll get a warning I'm sure you expect, given my above description:

`#warning "USB type is not set to XInput in boards menu! Using debug print - board will not behave as an XInput device"`

But as you can see (screenshot2.png), I don't seem to have the option to set the correct board type

[image]

mrebersv, 2025-06-07:
Alright, in order to clarify the steps and their order, I'm doing a fresh install.  

So, I uninstalled Teensyduino, ArduinoXInput, and Arduino.  I then did an install of Arduino and teensyduino and closed the arduino IDE on completion.  I went to `%AppData%\..\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\` and replaced the `boards.txt` file in there with the one provided in the ArduinoXInput_Teensy .zip release.  Then, in `%AppData%\..\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\cores\teensy4\`, I copied the files in the `teensy4\` directory from the ArduinoXInput_Teensy .zip release.  

Then, I booted the Arduino IDE, and I installed the ArduinoXInput library using the .zip method.  I then restarted the Arduino IDE.  I still do not have a "Teensy 4.0 w/XInput" option for boards, so I'm doing something wrong, but I'm not sure what.  Any help would be greatly appreciated.

dmadison, 2025-06-07:
There is no "Teensy 4.0 w/XInput". From the error message:

```
#warning "USB type is not set to XInput in boards menu! Using debug print - board will not behave as an XInput device"
```

Do you have an "XInput" option in the "USB type" menu? It should be the last one in the list.

---

From the `ArduinoXInput_Teensy` README:

>Version 2 of the IDE is reported to have issues with caching board files, which may prevent this project from loading or functioning properly. Consider using the version 1 IDE if you run into problems with installation.

I'm not sure if the caching issues have been resolved. If you're still running into problems, consider trying the version 1 IDE with the standalone Teensyduino installer.

mrebersv, 2025-06-07:
Thanks for the reply. I do not have the XInput option for USB type. I’ll
downgrade to arduino 1.x and try that.

closed 2025-06-10 by mrebersv